jkominek/piano-conversion #21: Check BOM against JLCPCB SMT parts list.

jkominek, 2020-12-22:
JLCPCB being one of the easier (and still reasonably priced) ways for the average person to get a board assembled, I'm trying to keep the parts used to things that are available via JLCPCB.

- [ ] Compile total BOM
- [ ] Check that JLCPCB has an appropriate corresponding part
- [ ] Document correspondence
- [ ] Build up BOM spreadsheets or something for use with JLCPCB? Haven't had them assemble anything, not sure how that works.

jkominek, 2021-01-13:
I went looking through JLCPCB's basic parts library for FETs for the main board. Looks like the relevant options are all SOT-23-3, so I'll need to tweak that for rev 1.

Here are SOT-23-3 FETs, and after brief inspection, the most likely candidates are marked with *

N-channel FETs: C8545(*), C20917
P-channel FETs: C10487, C15127(*), C8492

davidedelvento, 2021-11-10:
Hi @jkominek have you made any progress on this? And also with the CPL files? I would like to start ordering some boards, but I see only gerber files in https://github.com/jkominek/piano-conversion/releases so I'm not sure how to ask jlcpcb to do the SMT assembly without them.

jkominek, 2022-01-09:
No progress on this. I'm not sure it would even be useful at this point, with the widespread parts shortages. I was expecting to have to change some parts in order to make the board JLCPCB-assemble-able, which is fine, but doing it right now seems like an exercise in frustration.

jkominek, 2022-02-07:
I have started looking things up, in hopes of replacing some parts with ones which are more available / have more drop-in replacements, and also just assessing the situation. I used fewer distinct parts than I remembered!

https://github.com/jkominek/piano-conversion/wiki/Parts-Compatibility

So far it looks like the only totally unavailable parts are the microcontrollers, and the USB ESD protector. If there isn't anything else broadly available with a matching footprint, I'll talk to a real EE and design in something more generic.

davidedelvento, 2022-12-30:
> Compile total BOM. The first pass of this should probably ignore resistor values, 
> the ones in the schematics are very hand-wavy and don't strictly match what I've used.

Is this still the case? I see most resistor values are populated in the ADC board and they seem okay to me. Instead there are a number of missing capacitor values (some of which in the PCB appear to be connected nowhere, which probably should be a separate issues?)