jkominek/piano-conversion #33: Main board add RTC oscillator, battery header and appropriate Vbat circuitry

jkominek, 2021-01-20:
I'd like the ability to run the RTC on the main board, accurately. That'll let us track calendar time, so that we can properly time stamp files stored on the SD card.

Why do we want to time stamp files on the SD card? I'd like to be able to record everything played into MIDI files on the SD card, to duplicate the Pianoteq feature of remembering everything that's played. It's not a super high important feature, but it should be possible to at least squeeze the relevant stuff onto the PCB (with no downsides) even if it isn't populated.

See section 2.1.3 of AN4938. It looks like the STM32H743 can charge the battery a little bit? Do I need to put something in place to handle the possibility of a missing battery? Maybe a diode with a 1.5V drop from Vdd to Vbat? Or just some spot to solder a resistor between Vdd and Vbat?

So we need:

- [ ] 32.768kHz oscillator, reasonable accuracy
- [ ] Battery header. Polarized two-pin JST thing compatible with batteries like https://www.sparkfun.com/products/13852

davidedelvento, 2021-01-20:
That is a great feature to have!

jkominek, 2021-01-21:
Added in 2a38bf9

closed 2021-01-21 by jkominek