jkominek/piano-conversion #40: Pedal Board Rev 2

jkominek, 2021-02-01:
- [ ] The TRS tips, which goes to the ADC inputs, have a series resistors on them, but we should add some caps to those signals, near the ADC. This will help clean up the signal, and add ESD protection.

davidedelvento, 2022-02-16:
Perhaps add space for other kind of connectors as inputs, in addition to the TRS for external pedals. Some people might want to use a "regular" sensorboard to measure movements of something attached to salvaged acoustic pedals. The range of interesting movement for una corda, sostenuto and dampers is not very different from the hammers, so I'd expect three 1x boards working well for this use case, if this board is able to support them. No need for the more fancy stuff that you do for the hammers, IMHO.

jkominek, 2022-02-16:
Hmm, I'd imagined attaching mechanical pots to such things... somehow, and then cables with TRS plugs. The other thought I'd had was encoders, either mechanical or magnetic, which wouldn't work with this design.

But somebody using optical reflectance is also reasonable. https://www.sparkfun.com/products/15334 would cover the simple case, with appropriate firmware support on the main board. There wouldn't the same smarts that this version has.

I'll think about making an alternative "simple sensor board", which would have space for the same STM32G0, appropriate I2C ports, the IRQ line, and then headers to get at all the other pins, along with some optional analog tidbits one could use, or not. Like a quad op amp, and some good-but-not-great LED drivers.

davidedelvento, 2022-02-16:
> Hmm, I'd imagined attaching mechanical pots to such things... somehow, and then cables with TRS plugs. 

I see. I don't like mechanics, since all the leverage acting the pots need to be within well specifics ranges to operate correctly, but perhaps for just 2 or 3 as in this case it won't be so bad.

In my (*) view of what you are doing, I thought: an ADC measuring a potentiometer isn't that different from an ADC measuring the output of this thing https://www.sparkfun.com/products/11769 (which I am testing as a key sensor). Hence with limited work on your side, this very same board can support either the optical sensors, or the mechanically attached pots, or the commercial pedals based on pots.

(*): probably excessively simplified, as you have certainly noticed by my other comment about the op amp placement

jkominek, 2022-02-17:
> an ADC measuring a potentiometer isn't that different from an ADC measuring the output of this thing

Yeah, that's true. There's no reason I can't stick some headers on this exposing the ADC lines, and bring out the +3V3 rail a couple of times of convenience. I will do exactly that.

> I don't like mechanics, since all the leverage acting the pots need to be within well specifics ranges to operate correctly

That's a good point. Pedals would be moving pots at low RPM, so they'd probably hold up, but there'd need to be some consideration for alignment just to get it working in the first place.

jkominek, 2022-02-17:
Ok, added headers bringing out the ADC lines and some power, in 91a407f.

jkominek, 2022-04-24:
dc4713c marked as rev 2 and ordered

closed 2022-04-24 by jkominek