jkominek/piano-conversion #44: Sensor board PCB

jkominek, 2021-03-11:
I produced a bunch of variations on the sensor board very quickly, to account for the changes in my mechanical plan. I need to fix up the silk screen and labelling on all of them to make them less confusing. Some more test points would also be appropriate.

Additionally, I'd like to make a variation on some of them that uses the EAITRCA6 from Everlight. It's a similar, but much thinner sensor. Looks like it could be surface mounted, which would allow the sensor PCBs to rest flush against their support. That should eliminate flex when the hammer impacts.

davidedelvento, 2021-03-11:
Another idea on the topic of "avoid hammer impact", especially if the last 1mm is still too close to get any useful data. Have you tried if Plexiglas (or other transparent plastics) are transparent to these wavelengths? Usually these materials tend to be brittle, so they will need some support, but still it can be a viable alternative.

jkominek, 2021-03-13:
> Another idea on the topic of "avoid hammer impact", especially if the last 1mm is still too close to get any useful data. Have you tried if Plexiglas (or other transparent plastics) are transparent [...]

I have not, but it's worth giving a go at some point. A thin sheet of polycarbonate glued over top of my 3D printed support would help ensure heavily worn hammers can't slip into the sensor holes. And I'd expect polycarbonate to last forever under these forces.

I'll root around and see what I've got in my piles of materials, and maybe take a quick trip to Home Depot before we're all snowed in. :smile:

davidedelvento, 2021-03-24:
While playing with my hammers alignment, I've got more solid details on the ideas we floated on how to "screen" the sensors from hammer impact. 

The first is doable if you can make the 2040-centered-4x for a 30x60 extrusion (which should be pretty easy, since the latter is bigger). This option uses two https://us.misumi-ec.com/vona2/detail/110300466750/?ProductCode=HFSFIL5-3020-500 placed in front of the sensors leaving a slit for the sensor/LED to "see" the hammer. In this design the slit width is adjustable, and hopefully it's possible to make it thin enough to not let the hammers too much in, while at the same time large enough for the sensors/LED to see through. 

The other option, would use a 2080 with the sensor board mounted in the two innermost slots of the 80mm side and a number of polycarbonate sheets like https://us.misumi-ec.com/vona2/detail/110302696070/?ProductCode=PCTTA6H-70-70-5-F30-G60-N4 would be mounted on the two outermost slots. Kind of expensive though, so it gotta work well to be worth it.

I can make some sketches if you'd like.

jkominek, 2022-01-09:
It looks like you've already investigated this, but for completeness' sake I'll note that I haven't done anything with other sensors, but I think I do have a few EAITRCA6 laying around. Considering the shape of the leads, I'm not surprised that EasyEDA wouldn't have it, I don't think it could be used in automated assembly.

davidedelvento, 2022-01-09:
I have 10 'CA6 and I doing some tests with it. Already investigated is an overstatement, but I have done something, which I reported https://pianoclack.com/forum/d/243-diy-piano-sensors and in the PR #45 

I thought the 'CA6 had another option with regular surface mount seagull leads, but I can't find it at the moment. Alternatively, the QRE1113 could be an option: that comes in two slightly different forms, with either through hole or seagull leads. My results with it haven't been very good, but that was only one sensor and I might have mishandled or broken it, so it's worth investigating for the automated assembly.

jkominek, 2022-02-02:
3d1d8aee updates the 4x versions of the board for the parts of this that I still wanted to do, and introduces a version of the hammer sensor board with the sensors rotated 12 degrees, to make life easier with bass hammers.

And here's a PDF of the footprint for the QRE1113GR. I'm unclear if you had purchased any of them, but if you have, printing this at 100% and checking the correctness would be helpful. It's a non-standard footprint, so I had to make it from scratch, and I like to check parts against a printout before ordering PCBs. (The PCBs are cheap, I just hate wasting the turnaround time.)

[QRE1113GR.pdf](https://github.com/jkominek/piano-conversion/files/7983613/QRE1113GR.pdf)

davidedelvento, 2022-02-02:
I don't have any loose QRE1113GR at the moment, but I do have a couple of them soldered to the SparkFun board (see below). I could print the PDF on transparent paper (I should I have printable transparent plastic slides form the "good ole times", if not I can scavenge for the thinnest, semi-transparent paper I have in home, and hope the printer does not jam). With that, I should be able to check the match even if it's soldered to the board -- if you think it's useful -- since the sensor is thin, it may be better-than-nothing and good-enough.

Otherwise, I am making an order to DigiKey in the next couple of days and I can add one QRE1113GR.

Alternatively, we can do an all-software check with the data at https://github.com/sparkfun/QRE1113_Line_Sensor-Analog which is the board I have.

Yet another option: I have low-quality-but-usable setup to take super-close-up pictures, and I can take several of the QRE1113GR attached to the board, and placing close to it my "piano-caliper" (effectively a thinly marked ruler)

Let me know if any of these would be of any use (feel free to switch to email, if you think it's more efficient for this purpose)

jkominek, 2022-02-08:
I picked up some more QRE1113GR. The footprint matches just fine. 

I don't have the energy to keep track of what you're doing mechanically; is it sufficient for the new layouts to match the existing ones, just with this sensor replacing the CNY70?

davidedelvento, 2022-02-08:
> I picked up some more QRE1113GR. The footprint matches just fine.

Awesome, thanks!

> I don't have the energy to keep track of what you're doing mechanically; 
> is it sufficient for the new layouts to match the existing ones, just with this sensor replacing the CNY70?

I think mechanically I am doing exactly what you are doing:
- Place the same 40mm extrusions (with tracks at 20mm from each other) in front of the hammers. The hammers are aligned in such a way to hit the centerline between the tracks.
- will use a plastic "hammer stop rail" similar to yours in front of the sensors (obviously with smaller holes)
- for the 4x boards, the average distance between the centers of the hammers is 12.6mm and the section averages may go as high as 13.0mm and as low as 12.2mm (individual outlier pairs go even up to 15mm, but probably I should regulate them or use individual sensorboard for them) -- this variability is what I had in mind [here](https://pianoclack.com/forum/d/107-cybrid-a-diy-midi-controller-with-grand-piano-action/90)

I was not planning to use the 12 degrees boards, because I see my hammers do not all have the same orientation (which I have not measured). Heck some are even oriented in the opposite direction! If it cost you zero effort, make those too which might turn useful anyway to me or to someone else, but otherwise only the "straight" boards will suffice. 

Thanks a lot!

davidedelvento, 2022-06-23:
Have you placed the QRE1113GR (or EAITRCA6) footprints in kicad_mod format anywhere in the repo?
I'm trying to make PCB of the sensorboards with either of them and I can't find the appropriate footprint, but you must have made it already, since you posted the PDF in this thread.
Thanks!

jkominek, 2022-06-23:
Here's the one I made:

[QRE1113GR.zip](https://github.com/jkominek/piano-conversion/files/8967487/QRE1113GR.zip)

But I believe it is identical to the builtin footprint for the "OnSemi_CASE100CY", in the OptoDevice library. (Thought I mentioned that somewhere, but I can't find a record of it.)

davidedelvento, 2022-06-23:
Great thanks! Now I need to figure out the headers and mounting holes, which KiCad apparently cannot find (sigh)

```
Error: Cannot add J1 (footprint 'Pin_Headers:Pin_Header_Angled_1x10_Pitch2.54mm' not found).
Error: Cannot add H6 (footprint 'Mounting_Holes:MountingHole_3.2mm_M3' not found).
Error: Cannot add H5 (footprint 'Mounting_Holes:MountingHole_3.2mm_M3' not found).
Error: Cannot add H4 (footprint 'Mounting_Holes:MountingHole_3.2mm_M3' not found).
Error: Cannot add H3 (footprint 'Mounting_Holes:MountingHole_3.2mm_M3' not found).
Error: Cannot add H2 (footprint 'Mounting_Holes:MountingHole_3.2mm_M3' not found).
Error: Cannot add H1 (footprint 'Mounting_Holes:MountingHole_3.2mm_M3' not found).
```

davidedelvento, 2022-06-24:
If others have the same problems, I found the holes at https://gitlab.com/kicad/libraries/kicad-footprints/-/tree/master/MountingHole.pretty and the headers at https://raw.githubusercontent.com/KiCad/Pin_Headers.pretty/master/Pin_Header_Angled_1x10_Pitch2.54mm.kicad_mod -- note that the latter are from an older version of KiCad, because newer versions don't have that. Current v6 loads it ok, See https://raw.githubusercontent.com/KiCad/Pin_Headers.pretty/master/README.md for details. It seems to work, but I am not sure if there are some unintended consequences.