jkominek/piano-conversion #46: Scanning frequency analysis, per channel.

davidedelvento, 2022-01-08:
According to this paper http://mtg.upf.edu/mosart/papers/p35.pdf and this other one https://www.researchgate.net/publication/7603558_Touch_and_temporal_behavior_of_grand_piano_actions and how it discussed in some details at 
https://pianoclack.com/forum/d/289-scanning-speed-and-velocity-mapping-of-cybrid (and to less extent https://pianoclack.com/forum/d/243-diy-piano-sensors/33 ) it might be necessary to scan at 200kHz per channel.

jkominek, 2022-01-09:
Hmm. I had not thought about the timing in the context of the letoff. (And have not yet reviewed the provided materials enough to convince myself of the accuracy, but at glance it is concerning.)

As it stands the ADCs scan all the channels at the same rate. The firmware could, possibly, dynamically change the sampling rate. It could shift to sampling only one channel at a time when they're in the terminal stages of a >=forte hammer strike.

(There are other options, but that would be the "cheap" one.)

davidedelvento, 2022-01-09:
Yes, I have not convinced myself of what is really needed yet, but I wanted to put the discussion out there for you and others to see before anybody invests too much (time or money) in something which might be a dead (or subpar) end.

jkominek, 2022-01-09:
From a hardware design/posterity standpoint, my initial thought is:

1. Assume we only need this very high speed sending for hammers, not for key sticks/damper motion.
2. Lift the whole analog stage onto a daughterboard, intended to mount tightly to the ADC board.
3. Keep the TIA-configured opamps we've got, but then also add analog differentiators for 11 of the inputs. Those would be the hammer inputs.
4. Add two 16:1 analog switches/multiplexors. Each one would be connected to all of the differentiator outputs. So we'd be able to listen to any 2 of the hammer signals we wanted, differentiated.
5. The STM32H743 has two ADC inputs which are super high speed. Connect the analog switch outputs to those. Take all of the other analog inputs off of the corresponding ADC converter, and consolidate them on the remaining two other ADC converters. The sampling rate for those converters would drop further, but we'd just be using them to determine roughly when we're about to hit a "string" with a hammer, at which point we'd command the analog switch to feed us the differentiator output for that hammer. I'd have to recheck all the specs, but those high speed inputs should be able to get a high resolution measurement of the differentiator output, very quickly.

jkominek, 2022-01-12:
Pretty sure I can't actually squeeze on two more channels. So the above plan would have to be a single analog switch. And I think the firmware for flipping one of the high speed ADCs into that mode would be gnarly, at best.

Any thoughts on how fast the key stick/damper measurement needs to be? Seems to me that for measuring damper movement, we might be fine in the 100Hz range, but spatial resolution would be more valuable?

If I could drop 11 ADC channels off the STM32 onto some inexpensive external ADC. That would easily leave enough room on the STM32 ADCs and their input channels to accomplish the above. The high speed inputs could just stay in that mode, and probably hit >1MHz sampling, while the 11 always-on hammer sensing signals should be able to at least double their sample rates.

Reducing the number of ADC channels necessary on the main microcontroller dramatically eases the constraints on that part, as well. If it doesn't need >=22 ADC channels, there are other parts which are physically smaller, cheaper, and faster.

davidedelvento, 2022-01-13:
> Pretty sure I can't actually squeeze

I am not sure I understand the problem, so let me ask some questions. The speed limitation of many of the ST32 ADC channels, if my understanding is correct, come simply because of the extra resistance from the internal multiplexer, which calls for a longer settling time of the voltage. I thought that in your previous comment you were suggesting (among other things) to use a higher quality multiplexer to avoid that extra resistance and hence gain back that speed. Granted, the ADC has a maximum operating frequency, so multiplexing will (at best) divide that frequency by the number of channels, and that might be enough to rule this option out even with a hypothetical zero resistance multiplexer. But I am not sure that is what you are saying.

> Any thoughts on how fast the key stick/damper measurement needs to be? 

I am not sure how it *needs* to be, but key sticks move at about 5x slower speed compared to hammers, so I'd argue at least 5x slower sample rate will be ok. However they move about 1/5 of the distance, hence as you say we might need higher spatial resolution. On the other hand, many digital pianos totally ignore this issue, just use one bit spatial resolution (on/off) and according to those documents posted on pianoclack at 90us time resolution. Few people complain about that (I am one of the few), so I guess we can be "sloppy" there and it still should be ok. Hence:

> If I could drop 11 ADC channels off the STM32 onto some inexpensive external ADC. 

Tentatively, I'd say that it'll be just fine. We obviously need to run some tests before purchasing hundreds :-) of boards, but I don't anticipate problems.

> Reducing the number of ADC channels necessary on the main microcontroller dramatically eases the constraints on that part

I am not sure I follow you completely, so let me speculate. Are you suggesting a slow sampling for each note and each damper (and option for pedals), so 180 slow channels on an extra board? When those slow suckers detect enough of a movement the relevant input is routed to one of a small number (*) of very fast ADC? From a mechanical/pianist perspective, I think this will work fine, but I am not sure from an electronics pov: you'd need a 180-to-10 mux? You can't simply do 10 mux 18-to-1, because many if not all of the notes in flight could all be in a single one group of 18... Sure we can be smart with the geometry and multiplex notes that are impossible to be played at the same time, but it still looks very complicated to me. My understanding of multiplexing is limited though, so I might be totally wrong here.

(*) Pianists have 10 fingers, and often do not use all of them at the same time, so 10 channels will suffice, or perhaps 20 if we anticipate duets (or very acrobatic playing)

davidedelvento, 2022-01-13:
> The STM32H743 has two ADC inputs which are super high speed.

There are some higher end STM32 which have 3 ADCs (for a total of 36 channels). I almost bought the corresponding nucleo board a few weeks ago, before it went out of stock and so I settled for three RPi pico instead.

jkominek, 2022-01-13:
> I am not sure I understand the problem, so let me ask some questions. The speed limitation of many of the STM32 ADC channels...

Ah. Right, so there are a number of things tied up in there.

For starters, let me note that a very common issue with microcontrollers is that they've got more peripherals, and functionality, than can possibly be used simultaneously with the number of pins they've got. In part because the same die is used in many different packages; the STM32H743 that comes in a 208 pin package can use more of its peripherals simultaneously than can the 100 pin part. The 144 pin part I chose is, of course, in the middle.

The '743 has 3 AD converters, which have between them 36 input channels. I'm pretty sure that on the 144 pin part, a number of those channels come out on the same pin, reducing the practical count. Then some of them might come out on the pins of other things I need, as an example, the debugging pins. (That's an example, I'm not sure that's the case with this design.)

Next, the three ADCs. If you want the ADC to nicely capture samples for you at a fixed rate, and send them to you via DMA (very very important for performance), then all of the channels on that converter have to be captured at the same rate. So if I want to use the two very fast channels at a high sample rate, different from the slow sample rate, then they've got to be the only channels enabled for that converter. That leaves just two converters to do all of the slow sampling, for 22 channels, still.

Given the aforementioned pin limitations, I don't think I can get 11 channels onto each of the two remaining ADCs. Let's see, I'll pull up CubeMX and try to find an assignment...

Nope, can't get 11 channels on each of ADC1 and ADC2. I don't think I could even get 10 on both of them.

> There are some higher end STM32 which have 3 ADCs (for a total of 36 channels).

Yeah, the '743 is one of them. There might be Nucleos using a version of it with more pins, so they'd be more capable. I don't think ST makes any parts with more than 144 in the TQFP package. I don't really care to design with BGA. If you're going to have professional assembly done, they can do it, but it costs more and your yield will go down.

> I am not sure how it needs to be, but key sticks move at about 5x slower speed compared to hammers, so I'd argue at least 5x slower sample rate will be ok. However they move about 1/5 of the [...]

I'm willing to believe that the damper is lifted very quickly, but that doesn't seem very important to me. (I'm open to being convinced otherwise.) It's the return of the damper that seems important, and that's governed by gravity, right?.

I'd like more than 1 bit of damper position resolution, but I can't imagine needing 4 bits of resolution.

> > Reducing the number of ADC channels necessary on the main microcontroller dramatically eases the constraints on that part

> I am not sure I follow you completely, so let me speculate.

There I'm just noting that if we can reduce the number of ADC channels that have to come directly into the microcontroller, we can relax one of the constraints on choosing the microcontroller, and utilizing its resources. Anything which is happy with (hand waving, roughly) <=1kHz rates and 8 bit resolution, we could move onto an external ADC.

Definitely no 180:10 muxes. I don't think such a thing exists commercially.

Moving all of the ADC work onto a bunch of RP2040s is an option along those lines. One could imagine 6 RP2040s on every ADC board, sampling at 250kHz. They'd easily be able to implement the necessary location. Then a cheap many channel ADC for grabbing the slow signals. Or rearrange the whole system, so you have 4 boards with 11 RP2040s each, watching 22 hammers, and then pick up the dampers with some other board(s).

Though, a bunch of RP2040s raises its own issues.

> So I settled for three RPi pico instead, which for testing seem good enough (firmware, I dunno how much we can recycle, certainly some of the logic)

"Raw" firmware little to none, but the signal processing could certainly be reused, and that's the hardest part.

----

I really need to talk to one of my EE friends and see if they've got any ideas. I think I need to collect some more data from my existing setup, as well.

davidedelvento, 2022-01-13:
> For starters, let me note that a very common issue with microcontrollers is that they've got more peripherals, and functionality, than can possibly be used simultaneously with the number of pins they've got.

Yes, I did notice that and I told to myself: what the heck, they build these things with all of that stuff and then one can't use it all?? Of course I understand the economics of it, and also the size constraints that a smaller package may be more appropriate for some uses.

> Definitely no 180:10 muxes. I don't think such a thing exists commercially.
That's what I thought.

> Moving all of the ADC work onto a bunch of RP2040s is an option along those lines. One could imagine 6 RP2040s on every ADC board, sampling at 250kHz. They'd easily be able to implement the [...]

I see. So the RPi pico has not been a bad settling after all :)
So far there's not been shortage of those (yet?) so that's another incidental plus.
FWIW, I think the 9 bits of effective resolution are also adequate. I hope to have a 9-note (very roughly) working prototype in a weekend or two -- MIDI, calibration software et all. 

I think the thing that confused me was all this discussion about the slow vs fast ADCs. Sure we can offload the dampers to some other boards with slow ADCs, and perhaps it'd be more economical and/or easier to deal with -- however 4 boards with 11 RP2040 isn't very different from 8 boards with 11 RP2040 and have everything on the same hw and sw. Hence I assumed you wanted to reduce the number of fast ADCs by a on order of magnitude (which makes a big difference) rather than a factor of 2 (which may still be the right thing to do, and we may do it, but it's not a huge difference, at least in my mind).