jkominek/piano-conversion #47: ADC Board Rev 3

jkominek, 2022-01-15:
Place to record necessary changes for rev 3 of the board:

### Necessary changes

- [ ] Include footprints for low-pass filtering capacitors in the TIA circuits. Something about 10nF would probably suffice.

### Noncritical changes

### Possible improvements

- [ ] Consider moving to smaller JTAG headers.
- [ ] Consider pulling the entire analog stage onto a daughter board. There'd be additional cost in the board-to-board connectors, and of course the PCB, but otherwise I don't see any downside. There should be improved noise rejection between the analog and digital portions. The analog stage could probably be a two-layer board. Might be possible to make the ADC board two-layer at that point, as well.

jkominek, 2022-02-06:
Note to myself: We need to keep +3.3VA, as a requirement of the way the STM32 works. I want to bump Vce for the phototransistors up to at least 2V, preferably more. I don't want to run them off the input 5V. So we'll want analog voltage rails of 2V (vref), 3.3V (STM32 and opamps) and 4.5V (phototransistors). That'll leave us with 4.5V-2V=2.5V for Vce.

We'll say the conservative max for +4.5V will be 22 * ~5mA =~ 100mA. 2.0V will draw ~0mA, as it is just maintaining one of the high impedance inputs of an op amp. 3.3V will need to cover the STM32 and op amp circuitry. Maybe call it 50 to 100mA.

I'll abandon the interest in raising Vref; any improvement in "linearity" of the initial sensor signals is worth more.

davidedelvento, 2022-02-16:
> looking for ideas on improving/reducing electrical noise on the ADC board.

How about putting the op amp on the sensorboard? If I am not mistaken you are using parts which have 4 in one chip, so that's a perfect match for the 4x sensorboards.

jkominek, 2022-02-16:
> How about putting the op amp on the sensorboard?

That would require delivering a voltage source to the sensor boards.

I'm pretty sure it would also be worse; the current configuration was a deliberate design choice. The ADC input presents an extremely high impedance, so small amounts of stray charge can affect what it measures. By keeping the opamp output close to it, we make it easier for the op amp to control that voltage.

Whereas on the input side of the opamp, it shows ~0 impedance (because of the TIA configuration), so it would take a lot of stray charge to affect its inputs.

jkominek, 2022-04-24:
dc4713c marked as rev 3 and ordered

closed 2022-04-24 by jkominek