jkominek/piano-conversion #55: Cables

davidedelvento, 2022-07-25:
Would using these parts make the ADCboard to sensorboard cables easier to build?

Cable: https://www.digikey.com/en/products/detail/3m/HF100-40TP/3911407
Connector on the ADCboard side: https://www.amazon.com/dp/B00K2NTSJE/
Connector on the sensorboard side: regular jumper to be crimped, such as https://www.digikey.com/en/products/detail/harwin-inc/M20-1160046/3919438

This will cut in half the number of pins that need to be crimped (yay... but see below for even better) produce neatly organized cables (yay), and give some screening protection from noise by virtue of being twisted (yay).

I looked (not very hard) for single-row of stuff like https://www.amazon.com/dp/B0936Q6BDX/ which would eliminate the need of crimping.... or we can use two rows connectors rather than single row on the sensorboard too?

Now of course not everything will fit into the 4x arrangement (I'm planning to leave some of those sensors unconnected), so some cable-fu is still needed, but not as much as if we'd be wiring all 176 sensors (each with two cables, each with two ends) with a regular crimped connector. Am I missing something?

Note that if necessary the HF100-40TP cable can be ordered with custom lengths so we could make the distance board-sensor fully twisted and order an custom section of non-twisted part for ease of separating into the various sensorboards.

jkominek, 2022-07-25:
Huh. Something like that could work. It's 22 pairs, though, not 20. I'm under the impression that IDC crimping is sort of fiddly, so I'd assume I was going to have some wastage if I tried it.

> we can use two rows connectors rather than single row on the sensorboard too?

Certainly some sort of cable-fu would be required to make things line up.

There's probably some way to do something. I look forward to seeing what you come up with. 😃

davidedelvento, 2022-07-25:
According to https://blog.samtec.com/post/idc-or-crimp/ IDC are easy and reliable if you buy high quality parts.

They also have the 22 pairs (thanks for pointing it out) and the 8 in single row, so the parts are available.

They would even assemble it for you, and I assume that would be relatively inexpensive. However they default assembly configurator assumes you want the same connector on both ends (but can mount only one if desired). Most notably, they don't have the 3M HF100-40TP cable. 

As soon as I complete the PCB for the sensorboards, I'll start playing with this.

davidedelvento, 2022-07-26:
Which of these scenarios do you anticipate?
1. connect each pair of sensors (hammer and key) to the same ADC board, which implies
  * a slightly more complicated firmware is running on each ADC board, but they are all the same and they know everything about one note, so probably easier to debug and less work for the main board, _and, most relevantly for this issue_
  * each ADC board will have cables of different length to reach the differently placed hammer and key sensors _or_
2. connect hammer sensors to one ADC board, and key sensors to another which implies
  * a slightly simpler firmware running on each ADC board, but they are of two kinds (hammer and key) and they need to coordinate somewhat about each one note, so probably harder to debug and more work for the main board, _and, most relevantly for this issue_
  * each ADC board will have cables of one length to reach either the hammer *or* the key sensors

davidedelvento, 2022-08-08:
The use of 1-row connectors on the sensorboards and 2-row connectors on the ADC boards makes the ribbon-like approach probably too complicated (for people who don't know what I am talking about see [this](https://forum.digikey.com/t/ribbon-cable-pitch-vs-connector-pitch/248) article).

In hindsight (and for the people coming here in the future) I'd have made the sensorboards with 2-row connectors like the ADC boards (unfortunately they've just been manufactured, even if I don't have them in my hands yet as of now) -- much, much easier to do even if the number of terminations needed to be different (the ribbons could have been split, or more than one ribbon used per connector, which should be doable).

I'll keep experimenting and updating this issue, in the hope to find a good solution which does not involve tossing the sensorboards I have, nor manually crimping 176x2 pin terminations (this assumes that I do use IDC for the ADC side of the cables, otherwise it'll be 176x4 ugh)

davidedelvento, 2022-09-03:
More details on this later, but meanwhile, pictures

[image]
[image]

davidedelvento, 2022-10-09:
A quick report and a question.

I've made quite a bit of tests and I have not found anything that is "the perfect solution", however I've found a couple that can be adequate enough.

One is simply buying premade DuPont jumper cables of the appropriate length and gender -- one has only to twist them. One could buy them in ribbons and separate them: probably cheaper, but tedious and potentially prone to ruining some during the mindless separation (2 sensors per key, 2 cables per sensors that's almost 400 wires to separate). Alternatively one can buy individual cables. With this option one can use the individual housing for each connector, or the ones that can house more than one connector. A bit more complicated than I like to do the latter, but doable.

The other option involves IDC, however not the pre-twisted ones I looked at initially, which for a variety of reasons I did not find a good match. So this option too would involve a lot of manual separating and twisting. The advantage compared to the premade DuPont option is that one can use the appropriate connectors much more easily, and the appropriate cable length.

Yet another option would be to crimp DuPont jumper cables by hand. I find this too much work and discarded it too.

So in conclusion I would go for IDC or premade DuPont jumper cables, in either case twisting them by hand.
The question I have is about stray resistance. I've found that IDC connectors have a spurious resistance between nearby wires typically into the 40MOhm, and not infrequently as low as 10MOhm. I speculate that among the 400 needed wires one or two might even go as low as 1MOhm. Is this going to be a problem? Looking at https://github.com/jkominek/piano-conversion/wiki/Analog-Stage-Theory-of-Operation that resistance would be between emettitor and collector of phototransistors, either or both the *same* one, or a pair from two adjacent keys. As far as I can understand that circuit, this spurious resistance would just add a constant current which can be easily removed as part of the calibration, but I am not 100% sure hence my question.

davidedelvento, 2022-10-23:
Both crimping and IDC turned out to be too much of pain and time sink for me, so I settled on what I think is the easier and faster and does not require either. It does require a lot of twisting, and I'm investigating if I can reduce that, but otherwise, I'll leave with it.

I bought a bunch of pre-crimped cables at Pololu of all the various, appropriate lengths I needed. Available [here](https://www.pololu.com/category/72/female-female-wires-with-pre-crimped-terminals)

I also bought a bunch of connector [housing](https://www.pololu.com/category/70/crimp-connector-housings) in one and two rows. The former for the sensorboards, the latter for the ADC boards. The number of positions also were strategically chosen so that one final cable assembly connects two sensorboards (hammer and key) to is ADC board on a staggered mode: hammer-key-hammer-key-etc

Twisting the 12" cables which go to the sensorboards at the key is a bit tricky and I am still perfecting the technique, but I'm improving. One key to the success of the "perfect twist" is to not only twist the pair around each other, but to twist each cable on its own axis. When done properly, it results in a fast, stress-free, easy-to-bend pair, such as the orange gray on the picture below. If not done at all, it will result in a holy mess which will want to spring back in its untwisted position. When done so-so it will result in something that wants to make loops, has some internal stress, and is somewhat difficult to put in the position one may want, like the red-black and blue-yellow long pairs in the picture below. Note that the short pairs are much easier and forgiving to twist.

[image]

Probably it's clear and does not require a picture, but here is the detail of the 2-row ADC connector and its nearby connector for the sensorboard going to the hammer

[image]

And here is how I plan to assemble them, for now with a paper mock of the actual ADC boards:

[image]

Building a cable serving 4 notes, like the one shown in the above pictures, is comprised of 16 wires: 2 sensorboards (hammers and keys) each with 4 sensors; each sensor has two wires. It took me slightly more than 30 minutes, 15 of which twisting the 12-inch wires, and 10 minutes twisting the short cables (in the picture shown is 2 inches, but in other situation I will need 1 or 3 inches). It took me total 10 minutes to put the connectors inside their own housing: when the cable is twisted it's trickier than I thought.

If I don't improve my skills, that will take me 11 hours of work, which is reasonable but on the "I wish this was faster" side. I suspect I could reduce the time by 30-50% by improving my skills, but again that's still about 6h of twisting, which is too much. I am thinking of a LEGO-Technic basic jig to allow me to twist the cables of the right amount (on each own axis and around one another in the pair as I mentioned above). I will report here if I get any success at that.

davidedelvento, 2022-10-23:
https://www.youtube.com/watch?v=noDwhUSeec4